cvra/can-bootloader #54: Ignore extended CAN frames

nuft, 2015-03-15:
The bootloader client should discard extended CAN frames.
Otherwise it won't work with uavcan traffic during flashing.

antoinealb, 2015-03-15:
Agree. Can you do it ? I think a simple filter in the platform-specific code would do it ?

nuft, 2015-03-29:
I implemented the filter in the serial-can-bridge and master-firmware code.
To make use of this the client needs to be updated and set a filter for IDE flag to false with corresponding mask to must match (true). (see [can-bridge doc](https://github.com/cvra/serial-can-bridge/blob/b954b4a44d69670fb3dc0db545cd637e8f815cc1/README.markdown))

antoinealb, 2015-03-31:
I think we also need to do a change on the stm32f3 can interface, don't we ?

nuft, 2015-03-31:
No, this concerns only the CAN bridge communication between the PC and and the master board  - no changes needed for the motor control firmware or bootloader.
The changes on the master board firmware are the switch to the uavcan CAN driver.

antoinealb, 2015-03-31:
What I meant is that currently `can_interface_read_message` doesn't filter out extended frames.

nuft, 2015-03-31:
if you mean the bootloader code: no, the filter is already configured to only accept standard CAN frames.

antoinealb, 2015-04-01:
So I fixed it in d9d86d18debaba25e02bff50cd019061424d6908 in pure software. We must ignore those frames in software anyway, since some of them might fill the buffer before we configure the bridge.

closed 2015-04-01 by antoinealb