cvra/can-bootloader #76: GCC 6.3.1 generates too large binaries for some targets

nuft, 2017-10-02:
when building for motor-board:
```
arm-none-eabi/bin/ld: bootloader.elf section `.rodata' will not fit in region `FLASH_TEXT'
arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 76 bytes
```

antoinealb, 2017-10-03:
Interesting. Do you still have your previous GCC install to detect how much room we had before?

nuft, 2017-10-03:
With `gcc-arm-none-eabi-4_9-2014q4` I get 9996 bytes, which leaves 244 bytes room.

nuft, 2018-03-08:
Enabling link time optimization also fixes the issue (using gcc-arm-none-eabi-6-2017-q2).
You can do this by adding `-flto` to CFLAGS and LFLAGS in the Makefile.
I should fix this in a PR sometime...

antoinealb, 2018-03-08:
Is `-flto` stable yet ? Last time I checked it was full of weird bugs.

Anyway #78 should be a less risky fix IMHO.

SyrianSpock, 2018-03-08:
Solved by #78

closed 2018-03-08 by SyrianSpock